espressif/esp-eth-drivers #2: Version release?

stkw0, 2022-07-11:
I saw that there is a version release on idf component registry website but I don't see any tag on this repository. I think it will be nice to have a tag or a way to correlate which version corresponds to which commit.

Thank you

kostaond, 2022-09-05:
Hi @stkw0, good catch! I will add label.

kostaond, 2022-09-08:
I discussed the issue with my colleague and he had a good point: if we had multiple devices supported (which we want in the future), each would have its `idf_component.yml` with its specific version but the tag applies to the whole repo. Therefore it is probably not possible to do so and I will close the issue. Do you agree?

stkw0, 2022-09-09:
I see, so I should try to use idf component manager instead of just adding this repository as a submodule as I'm currently doing, right?

kostaond, 2022-09-09:
Yes, you are right. If you want to use it as is, you should use component manager. If you are planning to contribute or perform some tweaks, you need to use it as submodule.

stkw0, 2022-09-09:
okay. I will close the issue then

closed 2022-09-09 by stkw0

igrr, 2022-09-09:
> If you are planning to contribute or perform some tweaks, you need to use it as submodule.

Note that the IDF component manager still allows adding components from Git repositories, so it should work even if you need to fork the project at some point.
```yml
  ksz8863:
    git: https://github.com/your-fork/esp-eth-drivers.git
    path: ksz8863    # path inside the repository
    version: c469b510f71a46bcc993421f12e69472a5fcbe9a    # 40-character Git commit ID
```