espressif/esp-eth-drivers #8: I was thinking, can this be added to the arduino-esp32?

ChinaBurn, 2022-10-10:
If this switch chip can be supported, it will be very useful in daisy chain application scenarios

kostaond, 2022-10-19:
Thank you for the suggestion! Unfortunately, there is no short-term plan to add support of this chip to Arduino directly by Espressif. However, feel free to create a PR if you decided to do it by yourself or wait if somebody from community do so. Thank you for understanding.

rftafas, 2024-09-17:
Hi,

Indeed, there is no roadmap for that in the Arduino Core, thus, consider this won't go forward. On the bright side, this can be used along with the Arduino Component for ESP-IDF. We had a [very nice video during DevCon24](https://youtu.be/8w3MmsZBa9o?si=V_ddTRr2DwiM5K_A) about it. 

This can be closed.

kostaond, 2024-09-17:
@rftafas thank you for the suggested workaround. I'm closing the issue.

closed 2024-09-17 by kostaond