espressif/esp-eth-drivers #11: KSZ8863 I2C/SPI Read/Write Semaphore Not Released

drguthrie, 2023-02-09:
Thank you very much for the KSZ8863. It has been extremely helpful. While testing the driver over long periods I discovered an issue involving the I2C/SPI read/write locking.

```
    ESP_GOTO_ON_FALSE(bus_lock(KSZ8863_I2C_LOCK_TIMEOUT_MS), ESP_ERR_TIMEOUT, err, TAG, "I2C bus lock timeout");
    ESP_GOTO_ON_ERROR(i2c_master_cmd_begin(i2c_port, cmd, pdMS_TO_TICKS(KSZ8863_I2C_TIMEOUT_MS)), err, TAG,
                      "I2C master command begin error");
    bus_unlock();
err:
    i2c_cmd_link_delete(cmd);
    return ret;
```

If i2c_master_cmd_begin returns an error, the code jumps to err and does not release the lock. This code exists in both the read/write methods of the I2C and SPI.

I solved it like this:

```
    ESP_GOTO_ON_FALSE(bus_lock(KSZ8863_I2C_LOCK_TIMEOUT_MS), ESP_ERR_TIMEOUT, err, TAG, "I2C bus lock timeout");
    ESP_GOTO_ON_ERROR(i2c_master_cmd_begin(i2c_port, cmd, pdMS_TO_TICKS(KSZ8863_I2C_TIMEOUT_MS)), release, TAG,
                      "I2C master command begin error");
release:
	bus_unlock();
err:
    i2c_cmd_link_delete(cmd);
    return ret;
```

Thanks

kostaond, 2023-02-10:
@drguthrie thank you for reporting the bug and for providing potential solution. I'll have a look at it deeper. However, it would be very interesting to see what error the `i2c_master_cmd_begin` returns. Could you please investigate or provide steps to reproduce? The thing is it should never fail under normal circumstances.

drguthrie, 2023-02-10:
I will try to find out what error is returned.

I encountered the error two ways. First it occurred when I would leave my system sending sensor data over hours....when I returned it was occurring. Second I found it wold occur when stress testing the system by running a speed test through the switch while trying to send my data from the ESP.

Note that since putting in the fix to ensure that the lock is released, I have seen error messages printed out when running for hours. The difference now is that I get a single error instead of the continual errors produced once the lock cannot be obtained. Regardless of whether or not something can happen, it is never a good idea to have paths that lead to a lock not being released. My years coding that no matter how much I think something cannot happen....it always finds a way LOL.

drguthrie, 2023-02-15:
Quick (but not very helpful update). Despite a few attempts, I have not been able to capture the error. I am thinking maybe I had different task priorities or something similar when the issue occurred. I am still on the lookout but regardless ensuring that the lock is always released is a requirement of using them so the code will be more solid.

kostaond, 2023-02-20:
@drguthrie thank you for the update. I created PR with your suggested fix. I couldn't add you as reviewer but feel free to check it anyway.

closed 2023-03-03 by kostaond