espressif/esp-eth-drivers #16: ESP32 + KSZ8863 Daughter Board I2C Problem

AM0219350, 2023-08-18:
Hello,
I'm using this KSZ8863 ethernet driver to build my project. My hardware was assembled with ESP32 and AC320004-7(KSZ8863 Daughter Board). Now I can use SPI interface successfully and get the IP address from DHCP router. But i can't do the same thing with I2C interface. Here are what i have done and other information below. Thanks.

According to the document of AC320004-7. I wonder to use I2C interface.
1. R34 & R35 Replace with 4.7k
2. R28 & R29 [ Placed,DNP ] chose I2C Slave Mode
3. Revising the I2C speed with 100 khz. Because the 100 kHz it seems better than 400 kHz on SCL signal.
4. Revising the IDF menuconfig to set output of reference clock by ESP32 pin 17. 

IDF message:
[image]

I2C signal:
[image]

Reference:
1. AC320004-7 - KSZ8863 ETHERNET PHY DAUGHTER BOARD
  -https://www.microchip.com/en-us/development-tool/ac320004-7

Regards,
Ken.

kostaond, 2023-08-18:
It seems like sending command error, slave hasn't ACK the transfer.

> Revising the IDF menuconfig to set output of reference clock by ESP32 pin 17.

Is this the same configuration you used when KSZ8863 was in SPI communication mode?

AM0219350, 2023-08-18:
Hi Kostaond,
Thanks for your reply. I set the same configuration both SPI and I2C. They accept the input reference clock from ESP32 pin 17. 
Earlier time. I used the oscilloscope to check the signal. I found that my SCL amplitudes are very short. Sorry I didn't save that screenshot. But I draw it below. I used the simple_switch example. Just change the clock reference and conmunication mode. I think the signals are not similar to open-drain gpio.  Does the ESP32 needs to be specially set to open-drain when setting the I2C pin?

[image]

Warm regards,
Ken.

AM0219350, 2023-08-21:
Hi Kostaond,
I did some test for I2C siginal. And the other information for you. See the screenshots. Thanks.

1. Menuconfig
[image]
2. I2C open-drain circuit
[image]
3. I2C speed 400khz
[image]
4. I2C speed 100khz
[image]
5. ESP Version
[image]

Regeards,
Ken.

kostaond, 2023-08-21:
> I used the oscilloscope to check the signal. I found that my SCL amplitudes are very short.

Do you have probe attenuation properly configured for channel 2? It looks like 1/3 of 1V div, which is 0.33V which is 10 times less than expected.

AM0219350, 2023-08-22:
Hi Kostaond,
Sorry my it's fault. My oscilloscope probe is out of order. So that the I2C signal is correct. But I can't get response from KSZ8863(I2C slave mode). There are new information below. Is there any other way to check that the setting of KSZ8863 is correct? Thanks.

1. I2C signal from oscilloscope.
[image]

2. I2C signal from logic analyzer
[image]

3. Using ESP32 I2C tool to detect I2C addresses.
[image]

4. Screenshot 3 captured by logic analyzer
[image]

Regards.
Ken.

kostaond, 2023-08-22:
Using ESP32 I2C tool to detect I2C addresses is smart move. However, did you make sure that RMII REF CLK (50 MHz) was generated? If not, the KSZ8863 would be "dead".

It's really strange if it worked with SPI. I spent quite time thinking about the issue and I have no idea what could be wrong...

I would suggest the following:
1) Double check that changes required to switch the KSZ8863 to I2C client mode were properly done. For example, resistors were soldered reliably and no other components were affected. These are small parts and it may just happen desoldering one part, the neighbor component may fall due to heat...
2) Make sure that RMII CLK is present at REFCLKI_3 and X1.
3) Double check that SCL_MDC is connected to ESP32 GPIO 23 and SDA_MDIO is connected to ESP32 GPIO 18.

Regards,
Ondrej

AM0219350, 2023-08-22:
Hi Kostaond,
OK. I will check the I2C setting and contact the microchip for help. I will attach you the SPI interface status. There is a unknown io command. But the switch procedures still working successfully. Thank you.

1. ESP32 serial data
[image]

2. The possible position of the unknown IO command in the program.
[image]

3. Ping Test to 192.168.0.48
[image]

4. Hardware connection status. 
[image]
[image]

Regards,
Ken.

kostaond, 2023-08-22:
SPI looks good, therefore it is strange the I2C does not work. We have test board and use it in this mode without any issues.

Regarding `unknown io command`, this is really caused by 
```c
ret = esp_eth_ioctl(eth_handle, KSZ8863_ETH_CMD_G_PORT_NUM, &port_num);
```
Three Ethernet handles are registered, one of them is internal EMAC which is a "gateway" to the KSZ8863. The internal EMAC does not know `KSZ8863_ETH_CMD_G_PORT_NUM` option which is KSZ8863 specific. So the solution is either to make `unknown io command` error silent in the ESP-IDF or to add extra infrastructure to be able to distinguish whether Ethernet handle processed by even handler is KSZ8863 related or not. Feel free to create a new GH issue to address it.

AM0219350, 2023-08-23:
Hi Ondrej,
Finally it worked on I2C. I don't know why it works. According to the document of KSZ8863 module. It needs pull-up resistor on I2C pins for 4.7k. But in my situation. It only works with pull-up resistor 1k. At least I can continue my program job first. Next I will wait for our PCB to be made. Thank you again for your assistance.
Warm regards,
Ken.

1. KSZ8863 Daughter Board Information Sheet. page 5
[image]

kostaond, 2023-08-25:
@AM0219350, great to hear you were able to found the issue! Probably the impedance of longer wires could cause it? 

Could you please close the issue?

AM0219350, 2023-08-28:
Hi Ondrej,
In my experience. The modules that I used for I2C interface. They all use longer lines than this. The document suggested that I use a 4.7K resistor is also a reason. Keep me from trying other values. Thanks again for your assistance.
Regards,
Ken.

closed 2023-08-28 by AM0219350