espressif/esp-eth-drivers #17: [Feature Request] Support for RTL830xMB (RTL8304MB/RTL8306MB)

DjordjeMandic, 2023-09-27:
I see good use for KSZ8863 driver. Similar driver for RTL8306MB would be nice because it is low cost and available and supports 100BASE-FX.  Lot of 10/100M media converters use this chip. I see good use for integrated media converter and switch that also provides ethernet to ESP32. This is just cross post of [IDFFGH-9043](https://github.com/espressif/esp-idf/issues/10450) because it doesn't belong there.

kostaond, 2023-09-27:
@DjordjeMandic thanks for the suggestion. Actually, I've been also thinking about extending the offer of supported switch chips. I'll investigate.

kostaond, 2023-09-27:
@DjordjeMandic are you aware if any devbaord of RTL8306MB exists? It could potentially ease development otherwise custom board would have to be designed...

DjordjeMandic, 2023-09-27:
@kostaond it does not afaik. Currently biggest issue is that I just cannot find datasheet that contains description of registers for both RTL8304MB or RTL8306MB. 

 RTL8306E-CG seems to have well documented datasheet but its little bit different(2x RMII)
 https://www.lcsc.com/product-detail/Ethernet-ICs_Realtek-Semicon-RTL8306E-CG_C129792.html
 https://datasheet.lcsc.com/lcsc/1808291248_Realtek-Semicon-RTL8306E-CG_C129792.pdf

alexgoto, 2024-03-18:
@DjordjeMandic Hey. Were you unable to find the datasheet at the moment? I have same chip in one project and have the same problem. Link from lcsc.com not works for me, unfortunately. I use [this](https://www.datasheetq.com/pdf-view/RTL8309SB-Realtek) as a reference datasheet and tried to check out registers manually, with gdb. First 5-or-6 registers have been the same as reference, but another values turned out to be different

DjordjeMandic, 2024-03-19:
I put this project on the hold for now because of missing documentation and free time.

Alternative that I found promising is [Realtek-Unmanaged-Switch-Arduino-Library](https://github.com/shiroichiheisen/Realtek-Unmanaged-Switch-Arduino-Library). [Realtek_switch_hacking](https://github.com/libc0607/Realtek_switch_hacking) is good place to start. You might be able to find some useful info there.

Are you trying just to connect it and configure it as unmanaged or you're trying to use all of the features in it?

Gigabit ones have 8051 in them for managing(web interface and etc) but there is no RMII, only MII. (I don't know a lot about MII interface but I guess MII is pretty easy to implement but uses a lot of gpio. On other hand small cheap FPGA can bridge MII or even SMII from switch to RMII on ESP32)

alexgoto, 2024-03-19:
@DjordjeMandic thanks for reply. I'll study carefully into the links soon.I have found [this project](https://github.com/udif/kakaxi-project) which includes realtek switch driver today. It contains some interesting addresses, which does not described in realtek datasheets. Additionally it have defines for RTL8305MB.

alexgoto, 2024-05-03:
@DjordjeMandic Hey. I have done work with the switch. Driver code from https://github.com/lloves/RTL8309M-RK3288-KERNEL-3.4 has been working fine for my purpose. In my free time I could try to contribute my results to the project. But unfortunately I don't have experience with ESP-IDF previously and I isn't having any ESP's chip at this moment, for testing. My project target MCU was Gigadevice.

DjordjeMandic, 2024-05-03:
That's awesome, at least we have confirmed info so porting should be easy. Does vlan tagging and untagging work?

alexgoto, 2024-05-04:
> That's awesome, at least we have confirmed info so porting should be easy. Does vlan tagging and untagging work? 

Yes. SDK API pdf at the link contains description for vlan functions of the driver. I used the default vlan config where PVID is 1 for all ports

rikelme, 2024-12-21:
@alexgoto Hi there. We have a custom esp32 hw with the RTL8304MB chip on it. The ESP32 is connected to the RTL on 4th port (RMII Port), and there are 2 LAN connectors connected to the 1st and 2nd port. The first one would be used for the LAN connection, and the 2nd will allow other devices to connect to it and share the same LAN connection. In other words, a simple 2-port switch.
Would you be willing to help us make this work? Your contribution will remain in public domain, submitted here as a patch.

rikelme, 2024-12-22:
@DjordjeMandic maybe you?

alexgoto, 2024-12-22:
@rikelme Hi. I can try to help you, but only "blindly" because I don't have a board with esp32 and rtl3408 at the moment.

DjordjeMandic, 2024-12-22:
Hi @rikelme,

I currently only have ESP32 NodeMCU modules, so I don’t have the necessary hardware for this project. Ideally, I would need a PCB with the RTL chip and ESP32, or a breakout board for the RTL chip with at least two RJ45 ports. Test points for the SMI interface would also be useful for debugging with a logic analyzer if needed.

It seems that @alexgoto has already identified some good reference materials, which is a great starting point. From my perspective, the next step would be to try configuring the RTL chip to see if RMII works correctly. If that works, writing a driver would be the logical next step to simplify future implementations.

By the way, have you tried anything yourself so far, or have you primarily focused on developing the hardware without diving into the software part yet? This would help us understand the current state of your project and where we might assist.

While I don’t have experience with ESP-IDF, I’m confident that the many available examples would make it feasible to learn. I’d be happy to assist with development where I can, but my availability is limited due to my current studies. Without the hardware, I won’t be able to test or debug directly, but I can offer guidance as needed.

rikelme, 2024-12-22:
Thanks Djorde. I do have few hw prototypes. esp is flashed and running but
the ETH driver fails to initialize.

I have sent you an IM on Instagram.

kostaond, 2025-01-07:
If you don't need any management functionality of switch's ports, you can try to use https://github.com/espressif/esp-eth-drivers/blob/master/eth_dummy_phy/ component to connect to the switch via RMII. I presume the switch boots to some default configuration and then it's basically EMAC to EMAC communication. Could anybody who has hardware to give it a shot, please?

rikelme, 2025-01-07:
Thanks Kosta for your feedback. I do have the HW and will give this approach a shot as soon as we get back to office. I'll keep you posted.

kostaond, 2025-01-07:
I tested the idea with KSZ8863, it works like a charm :) Here's for your reference: https://github.com/espressif/esp-eth-drivers/tree/feature/mac_mac_switch/eth_dummy_phy/examples/generic_switch

crazy3min, 2025-08-14:
> I tested the idea with KSZ8863, it works like a charm :) Here's for your reference: https://github.com/espressif/esp-eth-drivers/tree/feature/mac_mac_switch/eth_dummy_phy/examples/generic_switch我用 KSZ8863 测试了这个想法，效果非常好:) 这里供您参考： https://github.com/espressif/esp-eth-drivers/tree/feature/mac_mac_switch/eth_dummy_phy/examples/generic_switch

Awesome, I made it happen.
But there is a problem. I configured UDP multicast, but cannot receive multicast data.
Send to the specified IP and the data can be received.

log：

D (2383) udp_audio: IP   : 192.168.68.100

D (2387) udp_audio: MASK : 255.255.255.0

D (2390) udp_audio: GW   : 192.168.68.1

D (2395) udp_audio: sock_udp[54] create Success
D (2399) udp_audio: sock_udp[55] create Success
D (2403) udp_audio: sock_udp[56] create Success
D (2407) udp_audio: sock_udp[54] bind Success
D (2411) udp_audio: sock_udp_recv[55] bind Success
D (2416) esp_netif_lwip: Multicast add filter IPv4: 224.0.2.6
D (2421) esp_eth.netif.netif_glue: Add filter MAC: 01:00:5e:00:02:06
I (2427) udp_audio: Joined multicast group 224.0.2.6 on socket[55]
D (2433) udp_audio: sock_udp_ctrl[56] bind Success
D (2438) esp_netif_lwip: Multicast add filter IPv4: 224.0.2.10
D (2443) esp_eth.netif.netif_glue: Add filter MAC: 01:00:5e:00:02:0a
I (2449) udp_audio: Joined multicast group 224.0.2.6 on socket[56]
I (2455) udp_audio: i2s rx start
D (2458) i2s_common: i2s rx channel enabled
D (2462) i2s_common: i2s tx channel enabled
D (4346) esp.emac: receive len= 60
D (5363) esp.emac: receive len= 60
D (6341) esp.emac: receive len= 60
D (7344) esp.emac: receive len= 60
D (8348) esp.emac: receive len= 60
D (9357) esp.emac: receive len= 60
D (10350) esp.emac: receive len= 60
D (11348) esp.emac: receive len= 60
D (12333) esp.emac: receive len= 60
D (13350) esp.emac: receive len= 60
D (14374) esp.emac: receive len= 60
D (15333) esp.emac: receive len= 60
D (20005) esp.emac: receive len= 60
D (21543) esp.emac: receive len= 61
D (27818) esp.emac: receive len= 194
D (122168) esp_netif_lwip: esp_netif_ip_lost_timer esp_netif:0x3ffbed78
D (122168) esp_netif_lwip: if0x3ffbed78 ip lost tmr: no need raise ip lost event
D (226872) esp.emac: receive len= 60
D (245285) esp.emac: receive len= 60
D (245287) esp.emac: receive len= 194
I (245288) udp_audio: Recv [64]bytes from [192.168.68.99:12345]
D (256510) esp.emac: receive len= 194
I (256511) udp_audio: Recv [64]bytes from [192.168.68.99:12345]

kostaond, 2025-08-14:
@crazy3min, I guess the multicast forwarding is not enabled in the switch. Check the datasheet (if you have such option).

crazy3min, 2025-08-14:
> [@crazy3min](https://github.com/crazy3min), I guess the multicast forwarding is not enabled in the switch. Check the datasheet (if you have such option).我猜交换机没有启用多播转发。请查看数据手册（如果有此选项）。

The multicast data was received, but the reception using **_recvfrom_** was unsuccessful. I don't know why.

This is data sent to the specified IP address and successfully received using **_recvfrom_**.
        D (2318705) esp.emac: receive len= 194
        I (2318706) udp_audio: Recv [152]bytes from [192.168.68.2:12345]

This is multicast data, and using **_recvfrom_** does not result in any response.
        D (2356553) esp.emac: receive len= 194

[image]

kostaond, 2025-08-14:
It seems your unicast and multicast socket sharing the same UPD port. Could you please try to bind each socket to different UDP port?

crazy3min, 2025-08-14:
> It seems your unicast and multicast socket sharing the same UPD port. Could you please try to bind each socket to different UDP port?您的单播和多播套接字似乎共享同一个 UDP 端口。您可以尝试将每个套接字绑定到不同的 UDP 端口吗？

The receiving socket is bindto port 60002.

Therefore, during the test, the target port is 60002, and unicast can obtain data.
Multicast cannot obtain data, but the receiving end prompts `D (2356553) esp.emac: receive len= 194`

```
/* 绑定接口 */
    if ((-1) != pstSelfInfo->sock_udp_recv)
    {
        /*---------------- 1. 准备本地地址结构体 ----------------*/
        struct sockaddr_in saddr = {0};                 /* 清零，避免脏数据 */
        saddr.sin_family        = PF_INET;                /* IPv4 协议族 */
        saddr.sin_port          = htons(60002); /* 将主机字节序的端口号转为网络字节序 */
        saddr.sin_addr.s_addr   = ip_info.ip.addr;        /* 绑定到本地网卡 IP（通常为 INADDR_ANY 或某网口地址） */

        /*---------------- 2. 绑定套接字到本地地址 ----------------*/
        err = bind(pstSelfInfo->sock_udp_recv, (struct sockaddr *)&saddr, sizeof(struct sockaddr_in));

        /*---------------- 3. 绑定失败处理 ----------------*/
        if (err < 0) {
            /* 打印错误码，方便调试 */
            ESP_LOGE(TAG, "sock_udp_recv Failed to bind socket. Error %d", errno);
            /* 关闭套接字并标记为无效，防止后续误用 */
            close(pstSelfInfo->sock_udp_recv);
            pstSelfInfo->sock_udp_recv = -1;
        }
        else {
            ESP_LOGD(TAG, "sock_udp_recv[%d] bind Success", pstSelfInfo->sock_udp_recv);
            /*---------------- 4. 加入指定的 IPv4 多播组 ----------------*/
            struct ip_mreq mreq = {0};

            /* 4.1 设置要加入的多播地址（例如 224.0.2.10） */
            inet_pton(AF_INET, "224.0.2.6", &mreq.imr_multiaddr.s_addr);

            /* 4.2 指定用哪张网卡接收（INADDR_ANY = 任意网卡；也可填 ip_info.ip.addr）*/
            mreq.imr_interface.s_addr = ip_info.ip.addr;// INADDR_ANY;

            int ret = setsockopt(pstSelfInfo->sock_udp_recv,
                                IPPROTO_IP,
                                IP_ADD_MEMBERSHIP,   /* 加入组播组 */
                                &mreq,
                                sizeof(mreq));

            if (ret < 0) {
                ESP_LOGE(TAG, "Failed to join multicast group %s, errno=%d",
                        UDP_AUDIO_RECV_MULTICAST_GROUP_ADDRESS, errno);
                close(pstSelfInfo->sock_udp_recv);
                pstSelfInfo->sock_udp_recv = -1;
            } else {
                ESP_LOGI(TAG, "Joined multicast group %s on socket[%d]",
                        UDP_AUDIO_RECV_MULTICAST_GROUP_ADDRESS, pstSelfInfo->sock_udp_recv);
            }
        }
    }
```

crazy3min, 2025-08-15:
> Could you please try to run [mulitcast example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/udp_multicast) on your hardware setup?您能否尝试在您的硬件设置上运行[多播示例 ](https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/udp_multicast)？

Thank you very much for your help, the issue is now resolved.

Problem description: ESP32 receives multicast data, but cannot extract the data using `recvfrom`.

Cause: Because ESP32 UDP binds to a fixed port (eg 60000), but the auxiliary test device sends data to a multicast address (destination port not 60000), `esp.emac: receive len = xx `will appear

Solution: Make sure the sending destination port and receiving port are consistent.