espressif/esp-eth-drivers #35: Unable to Obtain IP Address - Simple Switch Example

Ben-Rosenfeld-VPS, 2024-08-01:
Hi, 

I'm running the simple switch example and trying to follow the setup instructions. I have a PC connected to one port of the KSZ and a Netgear switch connected to the other port of the KSZ. In the logging, I am able to see the link up when I plug in the PC and the Netgear switch to the KSZ and ESP:

[image]

I am also able to see the Mac Addresses of all the devices:
[image]
[image]
[image]

However, I never see the Got IP Address log message:

[image]

I also used Wireshark on my PC but I never see these messages originated from the ESP:

[image]

Are there any suggestions you can make for us to debug this issue of not receiving an IP or transmitting data? We have connected a logic analyzer to the SPI bus of the KSZ and analyzed the data and it appears OK. Here is the configuration:

[image]

Any suggestions are appreciated. Thank you.

kostaond, 2024-08-08:
Could you please send full log?

Ben-Rosenfeld-VPS, 2024-08-15:
Just this you mean?

I (31) boot: ESP-IDF v5.2.1 2nd stage bootloader
I (31) boot: compile time Aug 14 2024 12:52:31
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.1
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=159f0h ( 88560) map
I (135) esp_image: segment 1: paddr=00025a18 vaddr=3ffb0000 size=02454h (  9300) load
I (138) esp_image: segment 2: paddr=00027e74 vaddr=40080000 size=081a4h ( 33188) load
I (153) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=388fch (231676) map
I (233) esp_image: segment 4: paddr=00068924 vaddr=400881a4 size=074c8h ( 29896) load
I (253) boot: Loaded app from partition at offset 0x10000
I (253) boot: Disabling RNG early entropy source...
I (264) cpu_start: Multicore app
I (273) cpu_start: Pro cpu start user code
I (273) cpu_start: cpu freq: 160000000 Hz
I (273) cpu_start: Application information:
I (276) cpu_start: Project name:     ksz8863_simple_switch
I (282) cpu_start: App version:      1
I (287) cpu_start: Compile time:     Aug 14 2024 12:52:17
I (293) cpu_start: ELF file SHA256:  e769e8900...
I (298) cpu_start: ESP-IDF:          v5.2.1
I (303) cpu_start: Min chip rev:     v0.0
I (308) cpu_start: Max chip rev:     v3.99
I (313) cpu_start: Chip rev:         v3.1
I (318) heap_init: Initializing. RAM available for dynamic allocation:
I (325) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (331) heap_init: At 3FFB3800 len 0002C800 (178 KiB): DRAM
I (337) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (343) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (350) heap_init: At 4008F66C len 00010994 (66 KiB): IRAM
I (357) spi_flash: detected chip: gd
I (360) spi_flash: flash io: dio
W (364) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
W (377) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (388) main_task: Started on CPU0
I (398) main_task: Calling app_main()
W (398) simple_switch_example: Simple Switch mode Example...

W (398) ksz8863_eth: SW reset resets all Global, MAC and PHY registers!
I (418) esp_eth.netif.netif_glue: d4:8a:fc:af:25:4f
I (418) esp_eth.netif.netif_glue: ethernet attached to netif
I (438) simple_switch_example: Ethernet Started
I (438) simple_switch_example: Dynamic MAC Table content:
I (438) simple_switch_example: valid entries 1
E (438) esp_eth: esp_eth_ioctl(496): unknown io command: 4109
I (438) simple_switch_example: port 1
I (448) simple_switch_example: Ethernet Link Up
I (448) simple_switch_example: 42 a5 70 93 28 47
I (458) simple_switch_example: Ethernet HW Addr d4:8a:fc:af:25:4f

I (468) simple_switch_example: Ethernet Started
I (478) simple_switch_example: Ethernet Started
I (438) main_task: Returned from app_main()
I (5468) simple_switch_example: Dynamic MAC Table content:
I (5468) simple_switch_example: valid entries 1
I (5468) simple_switch_example: port 1
I (5468) simple_switch_example: 42 a5 70 93 28 47

I (10478) simple_switch_example: Dynamic MAC Table content:
[... 171 lines cut ...]
I (130948) simple_switch_example: 94 a6 7e 78 5c c1
I (130958) simple_switch_example: port 1
I (130958) simple_switch_example: 94 a6 7e 78 5c c3

I (135968) simple_switch_example: Dynamic MAC Table content:
I (135968) simple_switch_example: valid entries 4
I (135968) simple_switch_example: port 1
I (135968) simple_switch_example: 34 56 fe 1c 0c 6b
I (135978) simple_switch_example: port 1
I (135978) simple_switch_example: 88 15 44 e0 e5 78
I (135988) simple_switch_example: port 1
I (135988) simple_switch_example: 94 a6 7e 78 5c c1
I (135998) simple_switch_example: port 1
I (135998) simple_switch_example: 94 a6 7e 78 5c c3

I (141008) simple_switch_example: Dynamic MAC Table content:
I (141008) simple_switch_example: valid entries 4
I (141008) simple_switch_example: port 1
I (141008) simple_switch_example: 34 56 fe 1c 0c 6b
I (141018) simple_switch_example: port 1
I (141018) simple_switch_example: 88 15 44 e0 e5 78
I (141028) simple_switch_example: port 1
I (141028) simple_switch_example: 94 a6 7e 78 5c c1
I (141038) simple_switch_example: port 1
I (141038) simple_switch_example: 94 a6 7e 78 5c c3

kostaond, 2024-08-16:
Try to increase log level to "debug" and observer for `"receive len= ` to see if you are able to receive data at least. You can also change log level directly at https://github.com/espressif/esp-idf/blob/release/v5.2/components/esp_eth/src/esp_eth_mac_esp.c#L297 to not pollute the log with other debug messages.

Ben-Rosenfeld-VPS, 2024-08-26:
Thank you for the guidance. We have narrowed the issue down to the following RMII Clock Selection bit needing to be set to 1 (internal) instead of the default (external):
[image]

We thought that would be this setting but with this set we don't believe Bit 3 in Register 0xC6 is being set?

[image]

This is the difference between the example schematic and our schematic:

[image]

Ben-Rosenfeld-VPS, 2024-08-26:
It seems we did just need to change that dropdown - is that all that is required from your expertise? Thanks!

kostaond, 2024-08-26:
So there is no issue in the driver if I understand it correctly. You just selected incorrect option at the start, am I right? The `CONFIG_EXAMPLE_P3_RMII_CLKI_INTERNAL` is default.

Ben-Rosenfeld-VPS, 2024-08-27:
We are testing now but will close this for now. The assumption is that it will work properly with the kconfig change.