espressif/esp-eth-drivers #40: KSZ8863 Ethernet Driver (beta) failing in I2C client mode

derrick-senva, 2024-09-24:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expect the I2C communications to work with the RMII clock generated by the KSZ8863 and for the Ethernet/RMII to work properly.

### Actual behavior (suspected bug)

The I2C communications fail to get ACK'd by the KSZ8863 when the RMII clock (REFCLKO_3) is enabled on it. The I2C traffic generated by the ESP32 also appears to sporadically generate invalid I2C traffic.

### Steps to reproduce the behavior

1. Configure the simple switch example from [https://components.espressif.com/components/espressif/ksz8863](https://components.espressif.com/components/espressif/ksz8863) for I2C client mode.
2. Wire the ESP32-PICO-DevKitM-2 to the [KSZ8863 Daughter Board](https://www.microchip.com/en-us/development-tool/ac320004-7)
3. Added an I2C command to the initialization code to configure the KSZ8863's RMII for MAC mode.
4. See the forum post here for full details: [https://esp32.com/viewtopic.php?f=13&t=41997&p=138041#p138041](https://esp32.com/viewtopic.php?f=13&t=41997&p=138041#p138041)

### Project release version

v0.2.6

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

CMD

### Additional context

This issue was originally posted on the forum with full details and screenshots of various analog/digital captures: [https://esp32.com/viewtopic.php?f=13&t=41997&p=138041#p138041](https://esp32.com/viewtopic.php?f=13&t=41997&p=138041#p138041).

derrick-senva, 2024-10-08:
This is what the 50 MHz RMII clock from the KSZ8863 looks like when we probe it at the IO0 (REF_CLK) header pin of the ESP32-PICO-DevKitM-2. Is this adequate?

If not, could this be the reason that we are not seeing activity on IO19 (EMAC_TXD0) and IO22 (EMAC_TXD1)? The serial monitor also says "simple_switch_example: L2 TAP write error, errno: 0".

[image]

kostaond, 2024-10-09:
The RMII REF CLK is 50 MHz square wave. You see something similar to sine wave but that might be expected. It depends on your oscilloscope bandwidth and you also need to disable 20 MHz filter (if it's there) on voltage input. If the oscilloscope bandwidth is low or filter is enabled, higher harmonics of the signal are filtered out and you cannot see the square wave. It's expected behavior, please try to search the web.

Anyway, have you tried to shorten wires as I suggested on forum? Did you double check the reference schematics with your setup. Did you double check the Kconfig configuration is in line with your HW setup? If so, please send me log of unmodified "simple switch" example.

derrick-senva, 2024-10-15:
> The RMII REF CLK is 50 MHz square wave. You see something similar to sine wave but that might be expected. It depends on your oscilloscope bandwidth and you also need to disable 20 MHz filter (if [...]

The scope is rated up to 200 MHz and the probes 350 MHz. The scope settings are configured for no bandwidth limit.

> Anyway, have you tried to shorten wires as I suggested on forum? Did you double check the reference schematics with your setup. Did you double check the Kconfig configuration is in line with your HW setup? If so, please send me log of unmodified "simple switch" example.

We tried shortening the wires and providing more ground connections between the two dev kits, but we can only shorten it so much. The Kconfig configuration is correct as far as I can tell. The I2C does work (without REFCLKO_3 enabled) and the RMII worked once as I was able to capture the Wireshark traffic. We don't have a schematic since it's just dev kits. I just have [this table showing our board-to-board connections and modifications](https://esp32.com/download/file.php?id=15250). We've reviewed it multiple times against reference schematics and the datasheets and cannot find anything that looks incorrect.

One of our engineers is going to prototype a custom PCB using the reference design and we'll try that (since the wires between dev kits are not ideal).

kostaond, 2024-10-29:
>  I'm also able to access the DHCP client on port 1 from my laptop

I'm not sure if I follow. Simple switch example workflow is to connect one to DCHP Server. You should got the IP address and be able to ping your device at that address. I don't see it in your logs... [Here](https://github.com/espressif/esp-eth-drivers/tree/master/ksz8863/examples/simple_switch) is example what should have seen.

Please send me schematics and your `sdkconfig.h` to ondrej.kosta@espressif.com. I'll have a look.

kostaond, 2024-10-29:
> The two things from the log that stand out are the I2C warning W (386) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h and
emac error E (447) esp.emac: emac_esp_custom_ioctl(276): unknown io command: 4109. The I2C is technically working fine in this example, so I'm assuming that isn't a serious issue, but what is the error?

Yeah, these are expected. I need to migrate to new I2C driver and suppress that io command error since it is expected.

closed 2024-11-08 by kostaond

lukedukeus, 2024-11-16:
I just ran into a very similar (possibly same) issue. I am using the simple switch example with no changes to the kconfig. My (messy) schematic, copied from [here](https://github.com/espressif/esp-eth-drivers/blob/master/ksz8863/docs/test_board/schematics/SCH_ESP32_Ethernet_TestBoard_For_KSZ8863RLL.pdf):
[image]

[image]

Output from terminal:
```
I (31) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (31) boot: compile time Nov 15 2024 17:01:16
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.1
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1653ch ( 91452) map
I (136) esp_image: segment 1: paddr=00026564 vaddr=3ffb0000 size=025e4h (  9700) load
I (139) esp_image: segment 2: paddr=00028b50 vaddr=40080000 size=074c8h ( 29896) load
I (153) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=397a0h (235424) map
I (234) esp_image: segment 4: paddr=000697c8 vaddr=400874c8 size=09344h ( 37700) load
I (257) boot: Loaded app from partition at offset 0x10000
I (257) boot: Disabling RNG early entropy source...
I (269) cpu_start: Multicore app
I (278) cpu_start: Pro cpu start user code
I (278) cpu_start: cpu freq: 160000000 Hz
I (278) app_init: Application information:
I (281) app_init: Project name:     ksz8863_simple_switch
I (287) app_init: App version:      042ab80
I (292) app_init: Compile time:     Nov 15 2024 17:01:01
I (298) app_init: ELF file SHA256:  fb835ddd4...
I (303) app_init: ESP-IDF:          v5.3.1
I (308) efuse_init: Min chip rev:     v0.0
I (312) efuse_init: Max chip rev:     v3.99
I (317) efuse_init: Chip rev:         v3.1
I (322) heap_init: Initializing. RAM available for dynamic allocation:
I (330) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (335) heap_init: At 3FFB39A0 len 0002C660 (177 KiB): DRAM
I (342) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (348) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (354) heap_init: At 4009080C len 0000F7F4 (61 KiB): IRAM
I (362) spi_flash: detected chip: gd
I (365) spi_flash: flash io: dio
W (369) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
W (382) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (393) main_task: Started on CPU0
I (403) main_task: Calling app_main()
W (403) simple_switch_example: Simple Switch mode Example...

W (403) ksz8863_eth: SW reset resets all Global, MAC and PHY registers!
E (413) ksz8863_ctrl_intf: ksz8863_i2c_write(77): I2C master command begin error
E (413) ksz8863_eth: ksz8863_sw_reset(142): write reset failed
E (423) simple_switch_example: ksz8863_board_specific_init(266): software reset failed
E (433) esp_eth: eth_on_state_changed(119): extra lowlevel init failed
E (443) esp.emac: emac_esp32_init(416): lowlevel init failed
E (443) esp_eth: esp_eth_driver_install(228): init mac failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400d8887
--- 0x400d8887: app_main at C:/Users/vanzw/OneDrive/Documents/Development/esp-eth-drivers/ksz8863/examples/simple_switch/main/simple_switch_main.c:306 (discriminator 1)

file: "./main/simple_switch_main.c" line 306
func: app_main
expression: esp_eth_driver_install(&host_config, &host_eth_handle)

abort() was called at PC 0x400884e3 on core 0
--- 0x400884e3: _esp_error_check_failed at C:/Users/vanzw/esp/v5.3.1/esp-idf/components/esp_system/esp_err.c:49
[... 43 lines cut ...]
I (139) esp_image: segment 2: paddr=00028b50 vaddr=40080000 size=074c8h ( 29896) load
I (153) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=397a0h (235424) map
I (234) esp_image: segment 4: paddr=000697c8 vaddr=400874c8 size=09344h ( 37700) load
I (257) boot: Loaded app from partition at offset 0x10000
I (257) boot: Disabling RNG early entropy source...
I (269) cpu_start: Multicore app
I (278) cpu_start: Pro cpu start user code
I (278) cpu_start: cpu freq: 160000000 Hz
I (278) app_init: Application information:
I (281) app_init: Project name:     ksz8863_simple_switch
I (287) app_init: App version:      042ab80
I (292) app_init: Compile time:     Nov 15 2024 17:01:01
```

And on the I2C line:
[image]

I'm seeing that my clocks are not square either:
[image]
[image]

I am using a cheap oscilliscope with a 100MHz sample rate, so it would not suprise me if this was not 100% accurate. From the error message I understand that the KSZ does not respond the the soft reset command (or any other command I try to send it). I have tried the other clock configurations listed in table 3-5 of the datasheet, but I get the same results:
[image]

@kostaond Can you provide any insight on what steps I could take to troubleshoot this issue? Thanks!

kostaond, 2024-11-26:
@lukedukeus, `E (443) esp_eth: esp_eth_driver_install(228): init mac failed` indicates problem with REF RMII CLK (missing CLK for internal EMAC). What is actual configuration of R1, R4, R6, R9 and R14 (or any other related to CLK configuration)?

lukedukeus, 2024-11-26:
@kostaond 

R4 and R6 on, R1, R9, and R14 are NC. SMTXD33=0, SMTXD32=0

I also ran into this when R1, R9, R14, and R4 is on and R59 is NC, SMTXD33=1, SMTXD32=1.

I would prefer to use internal clock, but at this point would like to get anything working.

kostaond, 2024-11-27:
In that case you should configure RMII CLK as internal in menuconfig and you should not observe `E (443) esp_eth: esp_eth_driver_install(228): init mac failed`. If you use GPIO, keep R59 and de-solder R92 (note these terminal resistors should be placed closed to ESP32).

Also just one note regarding using ESP32 internal CLK, you can use it only if you are not planning to use Ethernet and WiFi (or Bluetooth) at the same time (https://docs.espressif.com/projects/esp-chip-errata/en/latest/esp32/03-errata-description/index.html#clock-esp32-cannot-be-used-as-the-phy-clock-source-if-wi-fi-and-ethernet-are-used-at-the-same-time).

lukedukeus, 2024-11-28:
Thanks, I tried that configuration, but still ran into the same thing. I built 3 copies of the board, but still the same issue. Is there any hardware / design issue you can spot? I can see that the chip is generating a clock, but does not respond to any I2C commands, so I'm not sure what to change to get it to work.

kostaond, 2024-11-29:
Choose one CLK configuration and focus on that. I would recommend the one with external 25 MHz oscillator, KSZ8863 generating RMII CLK and ESP32 sinking it at GPIO0. Basically default configuration of examples. Then go step by step.
1) Make super simple app which just enables OSC and try if the KSZ8863 switches traffic at its ETH ports.
2) Move to Simple switch example. If you observe `E (443) esp_eth: esp_eth_driver_install(228): init mac failed`, focus on it. Measure both ends - CLK needs to be present at both KSZ8863 and ESP32 side. Try different values of terminal resistors. Triple check that all components are correctly assembled and correct values are used.

There might be problem with your PCB design - there might be improper impedance at GPIO0 since it's boot strap pin so there is more components connected to it and extra care needs to be taken. For example, we have a boards without programming circuit on PCB and external programmer needs to be connected. If improper wiring is used, signal integrity at GPIO0 is corrupted and EMAC doesn't work properly... If this is a case try to reconfigure and provide CLK from ESP32.

Try to follow https://resources.pcb.cadence.com/blog/2019-mii-and-rmii-routing-guidelines-for-ethernet.

Since this is obviously HW issue, I unfortunately cannot help you more... Good luck!

lukedukeus, 2024-11-30:
Thanks for the help. I will try redesigning the pcb.

kostaond, 2024-12-02:
@lukedukeus please stand by. I internally requested if our test board PCB Gerber files could be made public. It could help developers even for future... Let's wait if the request will be approved.

kostaond, 2024-12-02:
@lukedukeus here you go... https://github.com/espressif/esp-eth-drivers/pull/47