espressif/esp-eth-drivers #56: Add support for KSZ8863_SMI_MODE

rikelme, 2025-03-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

KSZ8863_SMI_MODE is currently not implemented. Due to this, hardware designs require either I2C or SPI, which unnecessarily increase the ESP32 pin usage for the KSZ8863 PHY configuration.

typedef enum {
    KSZ8863_I2C_MODE,
    KSZ8863_SPI_MODE,
    KSZ8863_SMI_MODE
} ksz8863_intf_mode_t;

### Use cases

Hardware designs without I2C & SPI control interface between ESP32 and KSZ8863.

### Additional context

The PHY control is required because at minimum the register 198 bit 3 has to be set to 1.

kostaond, 2025-03-18:
This is is unfortunately not possible, please see https://github.com/espressif/esp-eth-drivers/issues/20

Regarding the pin usage, the number of pins used is the same as SMI when you use I2C (2 pins).

rikelme, 2025-03-18:
Thanks Kosta for the feedback. Any thoughts on manually driving the SMI clock and sending the data conforming the custom Microchip SMI protocol ? In other words: a custom PHY implementation approach.

kostaond, 2025-03-18:
I'm sorry, I don't know and I don't see a reason to do so since it will always be necessary complicated. Maybe you could bend I2C since it's also consist of CLK and Data line...

If I remember correctly you can communicate with KSZ8863 even over ESP32 SMI BUT you will be able to access only basic PHY registers. In other words, you will not be able to manage the switch. If it's OK for you, you can use Generic PHY (`esp_eth_phy_new_generic`).

closed 2025-03-18 by kostaond

kostaond, 2025-03-18:
Or if your application does not need switch management, you have also option to use https://github.com/espressif/esp-eth-drivers/tree/feature/mac_mac_switch/eth_dummy_phy/examples/generic_switch. In such case, you don't need to connect the SMI at all and you save 2 pins (if pin utilization is an concern).