espressif/esp-eth-drivers #57: Add Support for Multicast Filtering within KSZ8863

Ben-Rosenfeld-VPS, 2025-03-20:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Support for Multicast Packet Filtering for KSZ8863

### Use cases

[image]

a.	Expect Device 1 KSZ to only pass through 239.255.0.2 multicast data to the ESP on the RMII
b.	Expect Device 2 KSZ to pass through 239.255.0.5 and 239.255.0.13 multicast data to the ESP on the RMII
c.	Expect Device 3 KSZ to only pass through 239.255.0.4 multicast data to the ESP on the RMII

### Additional context

The reason for this request is that currently when the data source is sending multicast data to a plethora of groups, the ESP32 is bogged down within the lwip stack:

[image]

And it ends up missing the UDP packets that are intended for it's multicast group. That's why we want the filtering to be enabled within the KSZ so that data does not even come across the RMII and use the ESP cycles.

kostaond, 2025-03-21:
@Ben-Rosenfeld-VPS thank you for the feature request. I'm actually currently working on adding an option to set multicast filters in EMACs based on join group content provided by lwIP. Would it be sufficient to filter multicast just in ESP32?

Ben-Rosenfeld-VPS, 2025-03-24:
> [@Ben-Rosenfeld-VPS](https://github.com/Ben-Rosenfeld-VPS) thank you for the feature request. I'm actually currently working on adding an option to set multicast filters in EMACs based on join group content provided by lwIP. Would it be sufficient to filter multicast just in ESP32?

I doubt filtering multicast just in ESP32 would solve our issue as the network has hundreds of multicast groups worth of data. The ESP is bogged down trying to process each packet. This is why the filtering really wants to be implemented within the KSZ.

kostaond, 2025-03-25:
OK, I'll try to put it a different way. ESP32 CPU would not see those packets since they would be filtered by its MAC module.

Ben-Rosenfeld-VPS, 2025-03-25:
@kostaond - that would potentially be viable. I'd like to test with it to ensure the solution works for our application. Is there a version of this implementation that I could try?

kostaond, 2025-03-26:
It's in progress. I have basic infrastructure at netif ready. However, it's not fully functional yet. I let you know once I have something to share.

kostaond, 2025-04-10:
@Ben-Rosenfeld-VPS the mcast update is currently in internal review. You can try to apply the following WIP git patch and test if it fulfills your use case: [mcast_filter.zip](https://github.com/user-attachments/files/19684962/mcast_filter.zip) 
Feedback is welcomed!

Ben-Rosenfeld-VPS, 2025-04-11:
@kostaond  - thank you very much. I'm away next week but will give it a try when I return (I doubt I'll have time to get to it today before I leave). Will provide feedback once it's up and running.

Ben-Rosenfeld-VPS, 2025-04-21:
Hi @kostaond - which version of the IDF were you using when creating this patch? I'm receiving errors while trying to apply on my side. I am using v5.2.1.

kostaond, 2025-04-22:
@Ben-Rosenfeld-VPS I use master, more specifically 8a98144711bbc40d5594b7d59d99ddbeca90fe45.

kostaond, 2025-04-22:
@Ben-Rosenfeld-VPS sorry maybe the patch file was corrupted. Please try the new one [mcast.patch.zip](https://github.com/user-attachments/files/19850402/mcast.patch.zip)

Apply the patch to full repository and then run `<PATH_TO_YOUR_IDF>/export.sh` inside your KSZ8863 project.

Ben-Rosenfeld-VPS, 2025-04-23:
Hi @kostaond - I tried it in a docker image - using the latest provided by espressif here (https://hub.docker.com/r/espressif/idf/tags?name=latest). It seems only an issue with one file?

[image]

kostaond, 2025-04-24:
@Ben-Rosenfeld-VPS please try with https://github.com/espressif/esp-idf/commits/8a98144711bbc40d5594b7d59d99ddbeca90fe45 as I mentioned before.

Ben-Rosenfeld-VPS, 2025-04-24:
@kostaond - perfect, thanks for the guidance - I've gotten it patched. 

Just so I understand correctly how to implement from my side - do I understand this correctly?

1. ETH_CMD_S_ALL_MULTICAST - this is TRUE when we want to receive all multicast data (no filtering); this is FALSE when we want for filtering to be enabled?
2. ETH_CMD_ADD_MAC_FILTER - when we have the filtering enabled and we ADD a MAC - is the MAC that is added one that passes through the filter and is provided to the upper layers? And if we don't ADD the MAC then it will be filtered OUT in hardware and not provided to the upper layers?

Thanks!

[image]

kostaond, 2025-04-24:
When you want to filter specific MAC addresses, `ETH_CMD_S_ALL_MULTICAST` must be set to `false` (default state). When you add address to filter table using `ETH_CMD_ADD_MAC_FILTER`, then it will be filtered in. If it is not added, such address will be filtered out as you mentioned.

You don't necessarily need to use `ETH_CMD_ADD_MAC_FILTER` directly, you can leave adding multicast MAC addresses on upper (IP) layer. For example, when you create a socket and you join to multicast group `setsockopt(sock, IPPROTO_IP, IP_ADD_MEMBERSHIP,  &imreq, sizeof(struct ip_mreq));`, this multicast address will be added to the filter via upper layers, see `examples/protocols/sockets/udp_multicast/main/udp_multicast_example_main.c` for more information.

Ben-Rosenfeld-VPS, 2025-04-24:
Got it - will test today. Thanks!

Ben-Rosenfeld-VPS, 2025-04-25:
@kostaond - this is great. It works for our application. Appreciate your help. Any idea when it'll be available in the main line?

closed 2025-04-28 by kostaond

kostaond, 2025-05-19:
@Ben-Rosenfeld-VPS, just FYI. It's been merged to master branch https://github.com/espressif/esp-idf/commit/eba9549bbc58988f4c959c1b8d5fa2636cbe3c45