espressif/esp-eth-drivers #59: enc28j60: IDF dependecy >=4.4 incorrect, would require 5.3

MastermindPanda, 2025-04-02:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Using the component with IDF versions >=4.4 works (in my case specifically 5.2.5)

### Actual behavior (suspected bug)

During CMake configuration, ESP driver component GPIO is not found

### Error logs or terminal output

```shell
CMake Error at C:/Espressif/frameworks/esp-idf-v5.2.5/tools/cmake/build.cmake:288 (message):
  Failed to resolve component 'esp_driver_gpio'.
```

### Steps to reproduce the behavior

1. Create e.g. 5.2.5 new project
2. Add dependency enc28j60 driver
3. Try to build

### Project release version

1.0.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

PowerShell

### Additional context

To my understanding, the CMake file already requires components in the "split" form introduced in IDF 5.3, see https://github.com/espressif/esp-eth-drivers/commit/521990810d24c301fb7556ec491e23832b15a78b#diff-d0a1bf59db9818ca37647723434aec902b4d433f710979ca77752fec5629e8c6R3

However, updating the required version to 5.3 would not a solution, at least for me. I'm still depending on 5.2.x, as this is the current version supported by the ESP AWS IoT libraries (https://github.com/espressif/esp-aws-iot)

igrr, 2025-04-02:
It could probably fixed similar to https://github.com/espressif/idf-extra-components/blob/2eaf9375478f8867204fcd9ae1ffd43662288b98/spi_nand_flash/CMakeLists.txt#L28-L32

kostaond, 2025-04-02:
@MastermindPanda thanks for reporting the issue! It will be fixed. In the mean time, please consider using workaround as suggested by @igrr or don't use the `enc28j60` from Component registry. `enc28j60` is part of the [ESP-IDF older releases ](https://github.com/espressif/esp-idf/tree/release/v5.2/examples/ethernet/enc28j60) so you can use it directly from the IDF.

closed 2025-04-22 by kostaond