espressif/esp-eth-drivers #71: ch390 not available 不能使用

qq547176052, 2025-06-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

[image]

[image]

### Actual behavior (suspected bug)

重启 

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

1
```
idf.py create-project-from-example "espressif/ethernet_init=0.5.0:common_examples/simple-ethernet"
```
set 

### Project release version

v5.4.1

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

Windows 10

### Shell

ZSH

qq547176052, 2025-06-23:
> What chip do you use? esp32s3

qq547176052, 2025-06-24:
> OK, we will try to reproduce. In the mean time, please try to double check that your SPI configuration is correct. If you use wires, use wires as short as possible (10 cm maximum). You can also try to decrease SPI frequency.

[image] 

[image]
Despite reducing the frequency and attempting to replace the SPI IO pin, the results still remain the same

kostaond, 2025-06-25:
My colleague tested this setup in our office and everything works as expected (we have our own CH390 SPI module). It's probably some problem with SPI signal integrity of your setup :/ You have quite long traces to the GPIO header. Or it could be the CH390 module. Is it something which is possible to buy? If so, send me a link to schematics. You can also try to measure SPI signals by oscilloscope. Check if you see clock, MOSI, MISO and CS signals to behave as expected.

SergeyKharenko, 2025-07-01:
@kostaond There is a certain possibility that WCH has secretly modified the VID and PID. If confirmed, I'll submit the patch as soon as possible. @qq547176052 Please tell us the actual value you have read. The code has already been replied to you in another repository.

kostaond, 2025-07-04:
Closing the issue as the issue cannot be reproduced as confirmed by @SergeyKharenko 

@qq547176052 please measure SPI signals using oscilloscope. Check if you see clock, MOSI, MISO and CS signals to behave as expected.

closed 2025-07-04 by kostaond