espressif/esp-eth-drivers #72: cannot add-dependency lan865x

RolliGrolli, 2025-07-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello,

I haven't to do so much with the add-dependency, it won't work with the lan865x.

When I call:
idf.py add-dependency lan867x                                                                                            
Executing action: add-dependency                                                                                                                                
NOTICE: Successfully added dependency "espressif/lan867x": "*" to component "main"                                                                              
NOTICE: If you want to make additional changes to the manifest file at path C:/Users/groll/esp/eth_spi_lan8650/main/idf_component.yml manually, please refer to the documentation: https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/manifest_file.html

This works well.

But the call to:
idf.py add-dependency lan865x                     
Executing action: add-dependency
ERROR: Component "espressif/lan865x" not found

I have tried just to copy the lan865x repo, but then the compiler cannot find some other files like "esp_eth_phy_lan86xx.h"

Maybe I just haven't switch to the correct branch or smthg. other.  Can anybody give me a hint how to build a project with the lan865x driver?

Thanks in advance:
Roland

kostaond, 2025-07-04:
Hi Roland, lan865x hasn't been released yet! Hence you cannot add the dependency. You need to clone the repo to your local machine. Then checkout to `feat/lan865x` branch. Now, you should be able to choose LAN865x in Menuconfig and be able to build common examples.

kostaond, 2025-07-18:
LAN865x has been release. Please use component manager.